Armchair-Heavy-Industries/Xol-Toolhead #4: Not enoug hotend cooling for Rapido UHF.

poppatchara, 2023-08-26:
Hi. I used Rapido UHF and old Mantis XOL2's HE mounts with no problem. But when I switched to XOL-Toolhead's new design with updated grills, I started getting more clogs. I decided to clip away some of the grills and the problems were gone.

I think the new grill design has too restricted airflow for the small 2015 fan.

DW-Tas, 2023-08-31:
Hi. Could you please confirm if your setup is using TAP or Xol-Carriage?
There is a future release planned that will remove gills from Rapido mounts and move to an open side similar to Dragon UHF mount: https://github.com/Armchair-Engineering/Xol-Toolhead/blob/main/STL/HE%20Mounts/Xol%20HE%20Mount%20-%20Dragon%20UHF%20-%20Sherpa%20Mini.stl

poppatchara, 2023-08-31:
Hi. It was Tap, vz-hex, Rapido UHF, bridge, with heatset fan. 

The fan was gdstime 12000rpm dual ball bearings, 2.45 cfm. Ran on max power.

My Tap is this clone cnc tap. 
[image]

DW-Tas, 2023-08-31:
TAP significantly reduces the space for exhaust airflow compared to Xol-Carriage, as it completely blocks the center-rear exhaust. We recommend the Delta 2510 cooling fan is it is capable of 4.3CFM and significantly outperforms any other tested 2510 fans.

I'm going to close this issue, as there is a release planned that addresses the airflow via gills, but I also recommend you move from TAP to xol-carriage and replace the GDS time fan with the Delta 2510 fan from the BOM to improve hotend cooling capacity.

closed 2023-08-31 by DW-Tas