Armchair-Heavy-Industries/Xol-Toolhead #8: Missing/wrong printing parts?

valepe, 2023-09-12:
Maybe it's my fault but looking for all parts to be printed there no evidence that optional parts have to be chosen for sherpa extruder.
Under sherpa mini I had to choose:
- STLs/optional_parts/[a]_housing_front_k_x1_rev15.STL instead of STLs/[a]_housing_front_x1_rev15.STL
- STLs/optional_parts/[a]_idler_arm_short_x1_rev15.STL instead of STLs/[a]_idler_arm_long_x1_rev15.STL

I made these choices based on xol 3d model, I suggest to add a note if choosing the sherpa mini extruder

DW-Tas, 2023-09-12:
Have added the following, including links to Annex GitHub for files.

[image]

closed 2023-09-12 by DW-Tas

patrezp, 2023-10-02:
annex just updated some STLs to rev16
[image]
this is the new url for idler arm
[https://github.com/Annex-Engineering/Sherpa_Mini-Extruder/blob/master/STLs/optional_parts/%5Ba%5D_idler_arm_short_x1_rev16.STL](https://github.com/Annex-Engineering/Sherpa_Mini-Extruder/blob/master/STLs/optional_parts/%5Ba%5D_idler_arm_short_x1_rev16.STL
)