Armchair-Heavy-Industries/Xol-Toolhead #12: PCB Klicky Holes Backwards

c0nsole26, 2023-10-08:
Not an issue per se, but caused me to reprint the part mirrored.  More of a heads up or a super gentle nudge perhaps.  I'm still building Xol, but am really excited (ty so much for the work done on it!)

**Issue:**  For the pcb klicky carriage mount STL, the holes appear to be flipped/mirrored.  When the pcb klicky mount is assembled on the carriage, there should be two holes on the left and one on the right if it is intended to work with the standard pcb klicky probe.

Pasting a ref link to the standard probe housing:  https://github.com/tanaes/whopping_Voron_mods/blob/main/pcb_klicky/STLs/probe-heatset.stl

DW-Tas, 2023-10-08:
It already is that way. The shaped part fits into the front piece of the carriage.
[image]

c0nsole26, 2023-10-08:
DW-Tas resolved on Discord.  I was using the incorrect front carriage (was using the Klicky NG one incorrectly).

closed 2023-10-08 by c0nsole26