Armchair-Heavy-Industries/Xol-Toolhead #14: Missing front (a) part for the Sherpa

MaikelKelder, 2023-11-10:
Cant seem to find the front part STL for the Sherpa Extruder. 
I would expect the STL in the Extruder folder, but not there, maybe I am missing it, but really tried finding it, believe me.
[image]

DW-Tas, 2023-11-10:
Sherpa-Mini is maintained by Annex Engineering. We have provided links to those parts in our documentation: https://github.com/Armchair-Engineering/Xol-Toolhead/blob/main/docs/printing.md

[image]

closed 2023-11-10 by DW-Tas

MaikelKelder, 2023-11-10:
Aaaah, Thank you, I looked at the default one, but noticed I needed one with a extra screw on the bottom. Thanks for the quick reply!!