Armchair-Heavy-Industries/Xol-Toolhead #23: Xol HE Mount - NF-Crazy_fanHS.stl feedback

YesterdaysCoffee, 2024-01-11:
All these mods I've done on my copy.
1) HS holes for hot-end fan could be 4-6mm deeper, otherwise plastic gets back filled into heat set, deeper holes will also allow for 20mm screws to be pulled snug.

2) The posts for the topmost fan mounts (when sitting on the build plate):  This 'wing' snaps easily, due to a smaller layer area at this point.
A small support web has been added, this web needs to be much larger.  On my copy I will widen the web in to the HE fan platform and out a good 10mm.  Perhaps out to flush with the part fan.

3) The bottom most HS post (oriented on the printer) would benefit by adding webbing to close the gap between main gap and horizontal HS socket.

DW-Tas, 2024-01-17:
Hi @YesterdaysCoffee, would you mind adding some screenshots to assist with your descriptions?

YesterdaysCoffee, 2024-01-18:
[image]
[image]

YesterdaysCoffee, 2024-01-18:
I am not suggesting the wire channel be moved or the wire splitter mounts be added.

DW-Tas, 2024-01-19:
1. Hole depth:
These holes are already extended to accept 20mm screws: <br />
[image]
<br/><br/>
2. Top "wings" breaking:
That is standard geometry on all Xol HE mounts and it doesn't have issues with "snapping easily". If it is breaking there for you, I recommend to print hotter and slower to improve layer adhesion.
<br/><br/>
3. Bottom heatset holes:
This is also standard geometry for Xol parts, but as the NF-crazy is a lower hotend than most, I will look to see if this can be done without increasing the toolhead weight. From a quick look, this may be possible withouth changing print perimiter lenght and may provide some bonus stability. Thanks for the idea :)

closed 2024-03-08 by DW-Tas