Armchair-Heavy-Industries/Xol-Toolhead #24: Faceplate - Standard - Rear - Low.stl

YesterdaysCoffee, 2024-01-11:
holes for fan screws are .5mm off center

DW-Tas, 2024-01-18:
Could you please add some more details? the rear faceplate has holes for hotend fan screws, and for the blower fan screws/heat sets.
Images would help if you can provide them here as well.
Thanks.

YesterdaysCoffee, 2024-01-18:
When the rear face plate is mounted with the blower fan screws, all HE fan screws do not line up.  And visa versa, when the rear face plate is mounted with the HE fan screws, all part blower fan screws do not line up.
My workaround was to enlarge the HE fan screw holes.

DW-Tas, 2024-01-19:
I can't reproduce the issue. Everything is lined up in the CAD. 
[image]

DW-Tas, 2024-03-08:
No reply, closing

closed 2024-03-08 by DW-Tas