Armchair-Heavy-Industries/Xol-Toolhead #29: Fairly Certain Delta Electronic's fan specs are erroneous for 	 ASB02505SHA-AY6B

DavidCC00, 2024-02-29:
Compare these two delta fans:

https://www.digikey.com/en/products/detail/delta-electronics/ASB02505SHA-AY6B/7491489  (5v from BOM)

https://www.digikey.com/en/products/detail/delta-electronics/AFB02512HHA-CF00/15970968 (Delta 12v otherwise identical)

There's only so many things you can do to improve a fan while keeping it's form factor the same.  

1.  Power (Watts or RPM)
2.  Change Fan blades
3.  Reduce friction (ie. maglev > ball bearing > sleeve/hydraulic.   Brushless > brushed motor)

The 12v fan beats the 5v one in both Watts (0.600 vs 0.960) and being ball bearing instead of sleeveless.   Delta's website doesn't specify fan blades, but they both appear to be the same large, flat 5 blade configurations (which is good for static pressure, airflow fans tend to have few skinny blades like a ceiling fan).   

In addition, the specs on the 5v fan rival that of much larger fans. 

I've emailed both digikey and Delta Electronics to find out, but unless something very odd is going on, I'm going to say it's an error.  I'll update here when I hear back.

lraithel15133, 2024-03-28:
@DavidCC00 any update on this? I am very curious.

DavidCC00, 2024-04-15:
> Douglas Hsia Douglas.Hsia@deltaww.com | Thu, Mar 7, 7:27 AM 
> 
> After checked with engineer, see below update fyi.
> 
> Reviewed the testing data in the factory, the current data is close to two model specifications.   So the specifications are OK.
> 
> Due to differences in the circuit design and bearing systems of these two fans, the efficiency of the fans is different.

I'm still really doubtful.  I've watched the fan showdown for years, have experience with brushless motors and watched lot of technical data on it (RCExplained does very technical dives),  etc.   The hydrostatic pressure rating on that, for an axial fan, is absolutely nuts.  Especially at ~30db.  Like, if it was 65db and spinning at 10krpm, I'd believe it.  For its size, that also seems like really good CFM.   Also, generally 12v fans are better than 5v fans just because they can generally produce more watts energy.  (Although, for whatever reason, maybe because they aren't as popular, 24v fans don't see much boost).  

However, it must be a pretty good fan regardless of accuracy of spec sheets, since I'd assume people would be reporting heatcreep issues, otherwise.

woodencliff, 2024-07-08:
If you comparing them side by side , the size of their fan hub is apparently different. ASB02505SHA-AY6B has the biggest disc area of its class, so it has best airflow! 
As for power consumption, I think there might be some volt conversion process in the higher volt model, which costs extra power.