Armchair-Heavy-Industries/Xol-Toolhead #33: LED Duct Interference Fit Issues

uosdwishpoom, 2024-03-21:
Both bottom PCF to LED Duct screws are an interference fit which is causing misalignment and ducts to split at the screw location.  Honeybadger fans from Fabreeko don’t fit the opening in the duct.  The fan and duct are .5mm from fitting along the long axis of the opening in the duct.

[image]

uosdwishpoom, 2024-03-21:
Fusion Filaments ABS 1.5 Uranium Yellow.  Shrinkage compensation 99.2.

uosdwishpoom, 2024-04-10:
Here is the solution I used and it worked great.
[image]

rcg94, 2024-04-22:
> Here is the solution I used and it worked great. [...]

Do you mind posting the file you made? Hoping to use it with my new honeybadger fans coming tomorrow