Armchair-Heavy-Industries/Xol-Toolhead #35: NeXtG UHF support?

DavidCC00, 2024-04-20:
The NeXtG UHF is apparently *almost* a rapido's dimensions, but without the taper up top.  I have one on order, and was hoping to build my printhead as a XOL.   Unfortunately, my understanding of CAD has been growing slowly, so I'm bringing this up.  (I can probably use tinkercad for personal use, but... tinkercad results in wonky files I'd hesitate to share with anyone).  

Here's where I got my info about the dimensions: 

https://www.printables.com/model/821299-drop-effectphaetus-next-g-uhf-stealth-burner-print

DW-Tas, 2024-04-20:
Please read [printing.md](/Armchair-Heavy-Industries/Xol-Toolhead/blob/main/docs/printing.md) from the docs. Next G part requirements are there. 

There is equivalent information in the pre-release version docs if you’re using that version.

DavidCC00, 2024-05-07:
Ah, good info.  I missed that, thank you.

closed 2024-05-07 by DavidCC00