Armchair-Heavy-Industries/Xol-Toolhead #36: Gap too large between Revo heat sink and cooling fan shroud causing heater problems

Urminme, 2024-04-30:
When using the standard Revo Heater core this is not an issue, but when using the Revo High temp Hotside which is just a standard heater block that fits a Revo nozzle I've noticed too much air from the Hotend cooling fan comes down through the large gap between fan shroud and the Revo Heatsink and blows over the actual heater block that it significantly impacts the heaters performance, ive done a few test where I've added some metal to this area to try and block the air from blowing down over the heater and its made a big difference, with it as is the heater is almost always @100% power trying to compensate and I've even had several prints fail as the heater cant keep up due to the excess cooling from the fan. Could you consider adding a version for the Revo(High Temp Hotside) that brings the hotend fan shroud closed to the heatsink to prevent excess air from flowing down over the top of the heater? Normally the heatsink does not get very warm at all and I don't think there is any issue of the heat sink melting the plastic especially with ASA or ABS, Toolhead's such as the stealth burner have the shroud basically touching the heatsink. 

I added a picture but its not the Revo hotend unfortunately however i circled the area of concern, I don't know if anyone else is even using this combo but I print a lot of PC, Nylon and ABS so I need the higher temps available with the Revo high temp hotside
[image]

DW-Tas, 2024-05-08:
I have made an update in the pre-release branch for revo-voron hotend mounts. Have a look at this one for an example. Let me know how it goes please. <https://github.com/Armchair-Heavy-Industries/Xol-Toolhead/blob/pre-release/STL/HE%20Mounts/Xol_HE_Mount-Revo-Voron_Fan.stl>

Urminme, 2024-05-08:
Thats much better, I love this toolhead and its been amazing in every other aspect. I appreciate your hard work. I will print this out today and check everything out, I'll let you know how it works, I dont know how many folks are using the Revo high temp hotside, but its very good.

closed 2024-05-20 by DW-Tas