Armchair-Heavy-Industries/Xol-Toolhead #37: BTT Eddy (Beacon Clone) Support in planing?

toscdesign, 2024-05-07:
Hi,
there ist a new Probe available.
The Bigtreetech Eddy.

This is a Beacon Clone with integrated Raspberry Pico and USB Connection:
https://github.com/bigtreetech/Eddy

It would bei nice to use it with the XOL Toolhead.
I have ordered one and would make myself available as a beta tester if needed.

DW-Tas, 2024-05-07:
So Xol isn't likely to support Eddie. There is a collision between Eddie and the X rail carriage. 
[image]

toscdesign, 2024-05-07:
Ah ok, thanks for the picture!

Is this the normal XOL or the longer Version for a Rapido UHF?
The long Version could have enough space, but I'm not sure.

Do we need the case from the Eddy?
As far as I know, it ist possible to use the pcb of the Eddy without it. The pcb itself hast mountingholes for screws.

This could be a way to use the Eddy with XOL

DW-Tas, 2024-05-07:
The PCB holes are in the same spot, and there is a JST connector at the top (in the gap shown in the picture). It would probably work with the UHF versions, but I am not going to make something that won't work with all versions.
The CAD for the existing probe modules is available if someone wants to make a UHF only usermod.

closed 2024-05-07 by DW-Tas