Armchair-Heavy-Industries/Xol-Toolhead #39: experimental cpap differences between ducts UHF and non UHF

Dario-Ciceri, 2024-06-08:
Hi, I checked the two models but I don't see any difference, is UHF a placeholder for the uhf version or does it refer to something else?
thanks

DW-Tas, 2024-06-08:
Hi. did you put them side by side in a slicer? 
[image]

Dario-Ciceri, 2024-06-08:
> Hi. did you put them side by side in a slicer? [...]

damn I didn't as I was using CAD and didn't notice... sorry...

closed 2024-06-08 by Dario-Ciceri