Armchair-Heavy-Industries/Xol-Toolhead #41: latest release seems dont support epdm belt

woodencliff, 2024-07-06:
While updating to latest release, I'm struggling to get epdm belts to fit in the new belt clips, only to find epdm belt is too thick, hope there will be a clip mod.

DW-Tas, 2024-07-13:
Print the clips with a couple of percent lower extrusion multiplier to make them a looser fit. 
Make sure to print lots of perimeters so they are basically solid. Also print the slow and hot for best strength.

aju3169, 2024-08-26:
I’m printing the belt clips on a BambuLab and the extrusion is set.  I can’t change the extrusion multiplier as far as I know.     It’s impossible to put the belt clips on for my edpm belts.

DW-Tas, 2024-08-26:
> I’m printing the belt clips on a BambuLab and the extrusion is set. I can’t change the extrusion multiplier as far as I know. It’s impossible to put the belt clips on for my edpm belts.

If Bambu slicer doesn't support EM (It should be under filament settings). You could use Orca Slicer.

Alternatively, fold up some sand paper and open the belt path a little that way.

aju3169, 2024-08-26:
Thank you - I sorted it out my using a very small file to open up the belt path.  Perfect now.

closed 2024-11-19 by DW-Tas