Armchair-Heavy-Industries/Xol-Toolhead #42: Belt lock for a Switchwire

rollduck, 2024-07-10:
Hi, 

Is there any small chance to adapt or make a new belt lock for Voron Switchwire something like this? https://www.printables.com/pl/model/773280-xol-belt-support-for-voron-switchwire
This one works very well, but maybe an official solution will be better? :) Please don't forget about this beautiful 3d printer :) 

Thanks in advance :)

skud52, 2024-07-14:
I've been working on a solution 

https://www.printables.com/model/943175-xol-switchwire-backplate

YoungHillionaire, 2024-07-31:
> I've been working on a solution
> 
> https://www.printables.com/model/943175-xol-switchwire-backplate

Thanks
Does this take the belt clips of the official SW or the XOL?