Armchair-Heavy-Industries/Xol-Toolhead #43: LGX Lite when?

GTreaper, 2024-07-13:
Hows the eta looking on can mount for the LGX Lite?

Thanks for the great work

MoppelMat, 2025-01-08:
I have the same question. I currently run SB with Tap and a LGX Lite want wanted to switch to this Toolhead. But as it seems there is no way to mount the EBB then. Since the LGX lite and the Tap is supported, I am a bit lost how to mount the Can board then. Or Can I just hotch potch my LDO NightHawk-SB on there "somehow"?

MoppelMat, 2025-01-08:
that looks promising!
thank you for your VERY swift reply.

sadly I have not done anything yet as I am still collecting options. But this seems like I can get started and see where I end now.
Thank you again.

closed 2025-06-04 by DW-Tas