Armchair-Heavy-Industries/Xol-Toolhead #48: Request: NF Crazy/Slice Mosquito hotend

notheotherben, 2024-09-09:
Hi there folks,

I noticed that you've got an NF Crazy Volcano hotend option, but didn't have a "standard length" variant for the standard NF Crazy/Mosquito/Mosquito Magnum. After hacking together a franken STL which combined the lower ducts from the Dragon SF and the upper mount from the NF Crazy, I can confirm that this works perfectly. Rather than submitting my (not very happy) STL, can I ask you to include this variant in a future release?

Thank you for all the hard work on Xol, it's been a treat to use.

DW-Tas, 2024-09-09:
Hi notheotherben,

We used to support NF Crazy with a standard block, but support for that was removed. The block is super wide and interferes with both airflow and lighting from LEDs.

The recommended approach is to use NF Crazy with an extender if you don't want to use the volcano block. I used the extender version quite a lot while developing the current release, and just like dragon with an extender, it has excellent airflow and lighting.

[image]

closed 2024-09-09 by DW-Tas

notheotherben, 2024-09-09:
Thank you, and apologies for missing that in the docs