Armchair-Heavy-Industries/Xol-Toolhead #50: Bambu HE mount interferes with TAP screws.

actualbigbobin, 2024-11-01:
top 2 screw heads interfere with the HE mount, it was also like that before the major update.

DW-Tas, 2024-11-01:
Can you please upload some pictures of what you mean? I don't understand the issue. Thanks.

actualbigbobin, 2024-11-01:
[image]
[image]

DW-Tas, 2024-11-01:
Thanks, that makes a lot more sense now. I've added this to the list.

actualbigbobin, 2024-11-01:
[Xol_HE_Mount-BambuLab_Fan.zip](https://github.com/user-attachments/files/17596184/Xol_HE_Mount-BambuLab_Fan.zip)
Here is my fix for it, I also extended the extruder mount hex nut hole through the other side because the nut got stuck once and it is impossible to remove it as is.